collin80/TeslaBMS #3: We are building a Tesla Powered car

jjwallace, 2017-07-05:
Hello, we are building an electric power car.  Which other boards have you tested on?  What would you suggest for a high-quality board?  I want to try interfacing with Tesla batteries this next week.

collin80, 2017-07-07:
Well, I've tested using an Arduino Due and a Teensy 3.6. I don't have experience outside those too. I did test a breadboard design with both of them so a purpose built board is not 100% necessary, just nice to have. There are a few things to keep in mind when searching for a solution to interface to the Tesla BMS sub-boards:

1. They're 5V so if you use a 2.5 or 3.3v board you will need level shifters. Both boards I used are 3.3v so I used logic AND gates as level buffers. 
2. The baud rate is 612500. Slow processors will either not be able to generate this high of a baud rate or will not be able to generate this exact baud rate properly. 612500 does not exactly divide into the clock rate of most processors. The processor on the Tesla main board is 24.5MHz which is how they generate such a screwy rate. Both the Due and Teensy can generate a suitably close baud rate. The Due has an 84Mhz clock and 84Mhz / 137 = 613138 baud which is close enough to work. Some processors do not have such free baud divisors. Technically 16Mhz / 26 would probably work though. 
3. A thousand amps tends to generate a lot of EMF so even if you didn't need logic buffers you might be best off using them as well as inductors and other noise cancelling devices. Battery packs are a noisy place to store electronics.

jjwallace, 2017-07-18:
Very interesting, this is great!

jorakarandas, 2017-10-23:
Hey Collin! Thanks for your contribution to the community! All the diy-ers appreciate it a lot.

 I encountered an issue with the Arduino Due and the Tesla BMS rev 2 and rev 3 boards. Arduino is sending the messages, but the boards don't respond back (from hardware point of view everything is fine, added lvl shifters).  Have tried 1 by 1 about 7 boards and another 8 which I installed in my car already in a daisy chain configuration as described by you.

What problems have you encountered during set up phase? Did the config work right out of the box? Was anything necessary from your side to make it work?
Thank you!!
[image]
[image]
[image]