collin80/TeslaBMS #4: Jumping voltage Measurements

tomdebree, 2017-07-21:
I am currently working on using the data read from the slave to control charging I noticed there is quite a jump in values. The jumps are around 0.05V, currently using a high level timer to reset once a certain threshold is past. 
Collin, did you notice the jumping of values too or is it possible to introduce a smoothing function over the values per cell?

collin80, 2017-07-23:
The resolution of the readings is a lot finer than 0.05v but if you're talking about full pack voltage then I suppose it is possible that 12 cells in series could add up to that much fluctuation. I hadn't noticed a constant 0.05 fluctuation but I suppose it does seem like the value wanders around a little bit.  It's certainly possible to introduce a layer of smoothing over the values. You could, for instance, take the previous 10 readings and average them. If you do this every 1 second then you'll get an average over the past 10 seconds. That might be OK for what you're doing or it might not. You can always read more rapidly and then average. If you need a 1 second value then you could read 8 per second and average over the 8. I don't know that such a thing will get added to this code base as it is meant mostly as a starting point for other projects and not as the absolute final code.

tomdebree, 2017-07-23:
I am using the commands:
bms.getAllVoltTemp();
bms.printPackDetails();

The values which are jumping are the cell voltages. these get put out roughly every 550ms

Attached a log from running these, all be it with some extra code to control charging. 
                 
[Tesla BMS voltage log.txt](https://github.com/collin80/TeslaBMS/files/1168440/Tesla.BMS.voltage.log.txt)

collin80, 2017-07-23:
Yes, I'm seeing it in your logs. It does appear to be about 0.05v. I'm thinking that it must be coming directly from the Tesla BMS module. I'd guess that their master board averages the voltages over multiple readings to make it more steady. Apparently the same thing will have to happen here. Sample more quickly and average the result. I'll try to do some more testing of this as well.

tomdebree, 2017-07-23:
If you can let me know how steady your readings are, I want to be sure it is not a wiring issue or something like that on my end.

collin80, 2017-07-25:
My readings are stable to within about 4 millivolts which seems to be a lot more stability that you're seeing.
I've attached a log from my pack. You can see that it sometimes fluctuates a little but, in general, readings are more stable than you're seeing. Perhaps your voltage source is shaky or you're picking up noise or your BMS modules are damaged. I can't tell for sure. But, hopefully having this baseline log file helps.

[TeslaBMS_Log.txt](https://github.com/collin80/TeslaBMS/files/1171632/TeslaBMS_Log.txt)

tomdebree, 2017-07-25:
Could you please test with turning the balancing function on?

tomdebree, 2017-07-25:
I can confirm it is indeed due to the balancing. Time to add another state to my state machine then.

closed 2017-07-25 by tomdebree