collin80/TeslaBMS #5: Tesla BMS reverse engineering

Pagode, 2017-10-30:
Hello collin80,

I am currently trying to connect Arduino Due with one Tesla Battery Package as part of a project for my study.
The Arduino is sending its messages but can't find the Tesla board. 
I am also not sure how to wire the fault output you described in your hackaday project. 
Hopefully you can give me any advice so that I can get further.

I am looking forward to hearing from you!

Reply:
[image]

Wiring:
red = 5V (Pin 1)
green = Ground (Pin 3)
yellow = TX (Pin 2)
white = RX (Pin 9)
black = fault (Pin 6) 
[image]

[image]

jorakarandas, 2017-10-30:
Hi Pagode,

You shouldnt connect directly the boards to arduino due. Grab a 5v to 3.3v
level shifter, like 74ahc1g08.

Tell us your progress after that.

Good luck,
J

danfador, 2018-06-10:
Hello, @jorakarandas do you succeeded with tesla BMS and Arduino, I have two model s module and I get an Arduino Due  + Molex connector, what else I have to boy to comunicate with Tesla BMS? 

One more question, without any external connection does testa BMS balance the cells ore not?

Thank you!

bstarr119, 2018-07-10:
Good Evening All,

I am having similar troubles as the original poster, unfortunately I cannot upload pictures from the mobile site. I have incorporated a level shifter with the due so that I do not over volt the tx rx ports. I will need to figure out something new to try but I am running low on ideas as I am new to CANbus and due ^^

NiallDarwin, 2019-08-21:
Hi @jorakarandas thanks for your suggestion of the level shifter. Please can you read this and let me know if I've understood correctly:
The ESP32 is a 3.3V chip & board so 5V could damage it.
The Tesla BMS boards are 5V.
This means that the Tesla board's Tx & Rx will be running at (up to) 5V and can therefore damage the ESP32. So using a level shifter means you can transmit the signal but drop voltage.
If all this is right I've got another question: Does the shifter cope with the baud rate OK? Is this a specification I need to look for when choosing a board?
Thank you :)

collin80, 2019-08-21:
Yes, the Tesla hardware runs at 5V but an Arduino Due or an ESP32 run at 3.3v instead. So you need a level shifter to go back and forth between the two voltage levels. What you need is a level shifter that can go fast enough. You have the baud rate of the Tesla hardware (625000) so what you need is a level shifter that can handle 1MHz digital signals. Cheaper shifters probably won't. But, plenty of options exist. And, in a pinch perhaps you've already got AND logic chips. Just feed the input to both inputs to the AND gate and output at the proper voltage. Logic gate chips tend to be plenty fast enough. You just have to make sure that the chips you're using will trigger at 3.3 and 5v even if the inputs are technically higher in voltage than the V+ rail on the chip.

NiallDarwin, 2019-08-24:
So I've now got  bi-directional board very like [this sparkfun offering](https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all) and have made something happen. Unfortunately it seems I'm only transmitting not receiving-see screenshot. Two things I'm not sure of* are:
1/ I'm getting my 5V from the 5V pin of the Due. It's getting it from my laptop. My meter shows about 4.9V which I'd expect to be enough. Thoughts?
2/ Does the shifter need any pull up/down on any of the inputs or outputs?
If you want to see the rig, [here's a little video](https://youtu.be/c_d3D_629mw).
Thanks :)
[image]
*Lots actually but these are the relevant ones ;)

wizkid057, 2019-08-29:
Was linked here by a friend.  I haven't really followed this project, as I developed my own software/hardware to run the BMBs quite a while back (I actually just reflash the 8051 using Tesla's own update protocol, reverse engineered, and run my own code there.)

Anyway, the digital isolator used between the BMS side and the external connector can operate at 2.5V to 5.5V... and can essentially act as a level shifter on its own.  Just power the external side from whatever voltage in that range would be useful to you.  The filtering before the isolator is geared towards 5V operation, and this is the voltage Tesla uses for the bus... but definitely not a solid requirement.  It'll give a few tenths of a volt loss most likely, but shouldn't drop things below the 2.5V minimum at 3.3V supply.

In fact, I'm actually currently running my own home setup with 36 of these in a chain, and it's running at 3.3V directly off of my board's 3.3V regulator (repurposed the board from another project... will eventually use 5V).  Realtime view: https://wk057.solar/bms.php

All of that said, I'm unsure what the benefit of level shifter would be exactly, unless the 3.3V rail isn't strong enough to provide the few mA per board needed to run the isolators.  You should just be able to power the whole thing from 3.3V and be done with it.  The BMB won't know any differently, since the isolated side is still powered by 5V.

collin80, 2019-08-29:
I'm sure you're right. We never investigated the situation that much. We saw that the Tesla main board uses 5v so we kept using that voltage and just added cheap AND gates to shift the signal level. But, it's good to know that this is not strictly necessary. So I suppose everyone else might as well just plain power it with 3.3v and be done.

BobbyBleacher, 2019-08-29:
Well, Niall is getting more than me out of it. When I start up it tells me starting up and factory settings reset, but how do I actually retrieve any useful data now that I'm connected?

BobbyBleacher, 2019-09-01:
Disregard, I've got this all connected, powered up, and operating.,The boards just aren't being recognized by the Due for some reason. I've tried the tx/rx pins as well as the cantx/rx with no success.

KeithMcdorman, 2019-09-02:
Worth noting that if you hooked up the due to 5v signals your signal pins may be toast. The due is not 5v tolerant. Been there done that.

BobbyBleacher, 2019-09-02:
Signal pins are still outputting 3.3v

NiallDarwin, 2019-09-21:
Alrighty, I made it work!
IT WORKS :)
The problem was simple. I was using the wrong serial pins on the Arduino. If you look closely there's three TX/RX ports. I'd gone in on the 0 pair but the board was using the 1s. Hopefully this helps.
We also found that as per lots of advice here the Tesla boards work fine on the 3.3V of the Arduino so no need for the level shifter.
To diagnose this a friend came over and brought his logic analyser. This enabled us to see what the various channels were doing and was much, much better than just blindly trying different leads in different pins. They're NZ$18 here in NZ so probably about US$10 and worth at least 100x more!

Summary of what works for me on my 2016 BMS boards:
Run Tesla boards on 3.3V from Arduino
Use RX1/TX1 pins
No need for level shifter
Baud Rate: 631578 (Haven't tried if this is necessarily the only one but it does work for me)

Thanks Collin for writing and informally supporting this stuff. Thanks everyone else for their support.

samsta, 2019-09-24:
Niall's friend with the logic analyser here.

In hindsight, we could have just looked at [config.h](https://github.com/collin80/TeslaBMS/blob/master/config.h#L10) which explicitly uses serial port 1 (not 0). Ah well ;-)

NiallDarwin, 2019-09-25:
I reckon this can be marked as closed now?

sogorman, 2019-09-28:
Wk057 (been following your work for a while, have chatted with you a few times ago about 5 years ago, big fan), @collin80 , and @NiallDarwin thanks for all the hard work on this project. I am trying to get my arduino to interface with the Tesla BMS and have not had much luck. I have tried running at both 612500 and 617647 baud rate with no response from the board. 

I took @wizkid057 advice and am running the board at 3.3V off the arduino header as to not require level shifting on the serial bus. 

One question for the three of you.... does the board need to be powered by the pack as well? not sure if the 6S leads from the cells power and subsystem on the module. 

I added some debug code to the setup function but here is all I am getting in the serial debugger. 

[image]

I am using Serial1 in config.h and the BMS is connected to RX1, and TX1 on the arduino. 

Here are some pics of my test bench... 

Anything jump out at you guys?

[image]
[image]
[image]

collin80, 2019-09-28:
As WK057 said, the power you give the board is powering the isolation chip. So, you can power that chip with 3.3 or 5v but it was meant for 5V so it probably works best with 5V if you can swing it. But, the BMS side is powered by the cells themselves. Ordinarily the current draw is quite low and everything wakes up when you send it traffic. But, you do need power at the cells in order for the 8051 processor and BMS chip to operate. So, that is your biggest hurdle.

NiallDarwin, 2019-09-29:
@sogorman Yes, the boards need to get power from the Tesla batteries. I've edited my summary above to include this important piece of info. You didn't miss it previously ;)

sogorman, 2019-10-03:
@NiallDarwin appreciate it and thanks to the entire group... she is now talking!

Nzo-Po1, 2023-07-04:
Bonjour @collin80 ,
J'ai besoin de ton aide concernant le code que tu utilises afin d'afficher la tension et la température du pack de batterie. En effet, j'utilise comme support la carte arduino DUE R3 qui fonctionne en 3,3V. Voila le code que j'utilise : 
[image]

Mes principales questions porte surtout sur le code car je ne sais pas où trouver la bilbliothèque SerialConsole.h. Je n'ai pas un niveau élevé en programmation mais il semble que ce soit une bibliothèque que tu ai fais toi même. Sais-tu où puis-je la trouver et comment l'ajouter aux bibliothèques de la carte ?
De plus, je ne comprends pas comment faire fonctionner cette ligne : SerialSpecialInit(USART0, 612500).
Suis-je censé ajouter une bibliothèque ou bien rajouter autre chose ? Pour être honnête, je ne sais pas à quoi corespond non plus les fonctions que tu as écrites comme "BMSModuleManager.h" ou bien "Logger.h". Tu les a écris en JAVA mais la carte Arduino n'et programmable qu'en C++ il me semble, c'est pourquoi je me sens un peu perdu :(

De plus, voici le câblage effectué, j'ai suivi les instructions que tu avais donné.
[image]

Merci d'avance de ton aide précieuse. ;)

collin80, 2023-07-05:
Well, of all the languages I can read, French is not one! ;) But, thankfully Google Translate exists. I believe you haven't properly downloaded all of the source code for TeslaBMS. The function "SerialSpecialInit" is found right in TeslaBMS.ino up near the top. The SerialConsole and other missing files are also right here in the repo so try downloading it all again. Lastly, make sure you have selected Arduino Due (Native Port) as the board you are compiling for. It's really easy to accidentally try to compile for a different board but this will not work.

Nzo-Po1, 2023-07-06:
Sorry @collin80, I forgot to translate ;) Indeed, I saw that I forgot to download all the libraries I need like due_can, wireEEprom ... So for now, the code compiles fine but I don't have the text in the serial monitor that I'm supposed to have ... Do you think you know why? I'll keep looking.
Anyway, thanks for your work, it helps me a lot :+1:

kveksen, 2023-09-23:
The tesla moduleboard is blinking but I just got this and nothing more.
What did I do wrong?

13:42:34.530 -> Starting up!
13:42:34.530 -> Started serial interface to BMS.
13:42:35.832 -> Resetting to factory defaults

[image]

Vistr169, 2025-01-24:
Hello @collin80,
Thank you so much for this programm, it work very well for old types of Tesla Batteries.
One question. Is it possible to make some changes at this code, that it work correct with Tesla Batteries 75kw, 100kw 2016-2021?
We had Tesla Model S Long Range 2021 with broken battery. Programm comunicate with BMB but not correct, and commands not the same.

evcepi, 2025-03-18:
Hi Enginers
i have problem with programing Arunio Due
somebody want help me private i pay for help?
I need to programing i need tool for see voltage cells and everthing?
Thank you