collin80/TeslaBMS #6: closing the main contactors inside the Tesla model-S battpack

TLS1000, 2018-09-21:
Hi, I'm trying to use the kit from EVTV, for using the complete model-S battery 'as is' without opening it.
Your code does a great job reading all cell voltages and so. BUT: the main internal contactors that connect the internal batteries to the power port of the pack don't want to close. Are these main contactors controlled via code or is it purely hardware?

collin80, 2018-09-25:
You'll have to ask Jack directly about that. I wasn't really involved in the whole pack control system. This project you've submitted the issue to is for controlling the modules inside the pack directly.

closed 2018-09-25 by collin80

TLS1000, 2018-09-25:
hmm...the collaboration with Jack is not working out :-) . We will try the kit on a second model-S battery to be sure, but I'm afraid we'll have to take the battery apart and go for 48V-setup.