collin80/TeslaBMS #7: ESP32 due code

martiinezz, 2019-07-28:
Collin - great work with the code. My question is if you're going to port this code to the latest ESP32 based board?

Thanks.

BobbyBleacher, 2019-09-16:
Is that why none of us can get this working? We've got a newer version of the board this was written for?

collin80, 2019-09-16:
Well, I did in fact port it over to the ESP32 for a project but that project didn't actually end up open source so it isn't on here. But, rest assured, it can work on an ESP32. I think really the biggest change to compile on an ESP32 is that it has no USB based serial so all the status output needs to go to a normal serial port instead and the serial port to communicate with the BMS slaves is initialized differently. 

Here's some quick advice for setting up the serial port to the tesla hardware:
HardwareSerial Serial2(1); //define second serial port for Tesla BMS board
Serial2.begin(BMS_BAUD, SERIAL_8N1, 4, 2); //rx pin is 4, tx pin is 2

Obviously BMS_BAUD should be one of the speeds, 612500 being a common one. There's a slightly faster one for newer Tesla hardware, 631578. I think we had better luck just using 631578 all the time. I don't think the ESP32 can quite generate that exact rate but it gets close enough.

So, no, I don't believe the ESP32 is available anywhere but really you just redirect all serial console output to Serial and all BMS traffic to Serial2 - initialized as above.

BobbyBleacher, 2019-09-17:
I'm staring at this code trying to figure out how to make it do just that. I'm mostly unfamiliar with all of this, so most of what I'm trying results in failure to compile or something I don't even want to go try after it does compile because it simply doesn't make sense. 

So the Due, which does have SerialUSB, needs to NOT use that and initialize two separate serials (looks like HardwareSerial doesn't need the initialization code you put there), and the Serial2.begin statement provided has two too many arguments. 

 What was this originally written for if not the Due?

collin80, 2019-09-17:
No, the code as provided should have worked on the Due. That's what it was written for. I was talking about the ESP32. In order to compile for ESP32 you need the special serial init I posted and you can't use SerialUSB since ESP32 has no SerialUSB - Due does.

BobbyBleacher, 2019-09-17:
Oh... Yes, it definitely doesn't work on the Due as noted in Niall's video. I'm getting the same results as him. LPG on the Tesla forums also had the same experience. 
[image]

collin80, 2019-09-17:
What is interesting about your picture is that you are getting the exact same bytes returning as you sent. That isn't really what it should be doing. The Tesla module should be catching the bytes that you send and be broadcasting but with changes. 7F is a broadcast and 0 if I remember correctly is used to number the modules automatically. It shouldn't be directly passing those bytes back. Something seems odd there. You should make sure you don't have a direct loop between TX and RX. Does your first tesla module in line have LEDs flashing on the board when it gets traffic? Near the bottom you should see flashing upon traffic. This can help to verify that it is receiving data bytes.

BobbyBleacher, 2019-09-17:
The light on the BMB comes on only during power up, but that's it. If you're interested in troubleshooting (and I definitely am) we can chat elsewhere. I have Whatsapp, discord, slack, etc.

collin80, 2019-09-17:
Yeah, sounds like you've got some major comm issues then. Every single time the code tries to send something to the Tesla modules you should get a flash of light at the LED. So, things are not getting through properly. I have slack and I think discord. We can move this out of the github issues and to email, then perhaps to one of the other services from there. kidderc at google's email service.

CZal, 2019-09-17:
Code works for me... I think there is an issue with your wiring to the modules....

Btw, Great work Colin...

BobbyBleacher, 2019-09-17:
Care to share some knowledge? There are several of us that can't get it working and it does not appear to be the wiring. Would love to see photos/videos of your setup.

martiinezz, 2020-01-19:
@BobbyBleacher So what was your issue?

twanek, 2021-02-19:
hello!

thank you sharing this great work! i also would like to port this code to esp32. anyone can help me, where to start? unfortunately, i do not have a due, so i have to begin directly with the esp32... 

best regards!

collin80, 2021-02-24:
I gave the relevant advice up above. I don't know for sure what else you'd want to know. On the ESP32 your user facing output should be sent to Serial instead of SerialUSB and you need to initialize the second serial port that talks to the modules as I specified above. Other than that, both chips support Arduino so once you install the Arduino support for ESP32 you should only have to make those two changes and it should work.

MohamedGaberZidan, 2022-12-05:
> I gave the relevant advice up above. I don't know for sure what else you'd want to know. On the ESP32 your user facing output should be sent to Serial instead of SerialUSB and you need to initialize [...]

Hello,
when I did this I had some problems with due wire library, could you tell me what I should do?. Thanks in advance.
best regards.