collin80/TeslaBMS #8: balancing?

Energyfuzzi, 2019-08-19:
Hi, great job!
I try to implement it on a Volkswagen Beetle with 5 Tesla Bricks, looks good!
While checking the balancing, I found that there is something strange inside your code:
You start balancing at 3.9V, and stop when the cellVoltage is below this Voltage minus a balancing hysteresis. 
There are 2 problems I see, because "balancing(i)" seems to mean: burn the energy at the resistors at cell(i)! 
1) So the Resistors from each cell burn energy every time the Voltage is higher than 3.9 V unless it is below 3.86V, even if the voltage should be at 4.1V.
2) During balancing the Voltagemeasurement does not work properly, because of the structure of the electronic.

I would suggest starting balancing, if the difference to the neighbour cell is to high, maybe via the difference between avgCellVoltage and cellVoltage(i), or just by comparing, and a minimum voltage is guaranteed (as already done). 
Then the balancing should last 2-5 min, should stop, then after a short time start the measurement and the balancing again.
What do you think?

collin80, 2019-08-20:
You're pretty much right all the way around. The balancing stuff found in the public code is only to be used as an example of how one might be able to turn on balancing. It certainly isn't a very good algorithm. So, yes, I would recommend you implement something better.

Energyfuzzi, 2019-08-20:
thanks, I´ll change it for my needs on the Volkswagen beetle. 
And by the way: thanks once more for finding out, how to communicate with the original TeslaBrickElectronic. It really helps to use the original Brick instead of using a custom BMS!
Has there been something new about the communication in the last 1-2 years (since you posted the code?) All the best from Austria!