collin80/TeslaBMS #9: BMS board revisions? What should I look for?

baysinger, 2019-09-24:
If I were to purchase a module, is there any particular BMS revisions I should look for? Or are all the Model S modules compatible with this project?

baysinger, 2019-10-18:
To partially answer my own question, it does indeed work with whatever this is:

[image]

collin80, 2019-10-18:
Sorry, I meant to answer this long ago but completely forgot about it. Any of the Model S battery revisions should work. It appeared that they changed serial speed between revisions but only slightly and the code can support both. To my knowledge there are two main revisions of the BMS boards. One is larger than the other. But, either way they speak the same protocol and the only communication difference is a slight speed change. The old ones were 612500 and the new one something like 631000 (don't remember exactly). Other than that they work the same.

baysinger, 2019-10-22:
Thanks. BTW, I left the baud rate alone (617647) and it's working fine. Not sure where that number comes from, though. Close enough to either speed to work, maybe?

closed 2019-10-22 by baysinger