collin80/TeslaBMS #11: Can't get more than 14 modules to register

BobbyBleacher, 2019-12-02:
And it looks like the code is right. It should theoretically go to 62 without any issues. Any else experience anything similar?

collin80, 2019-12-03:
Yes, the theory is 62 as that's how many addresses can be assigned. I'm not aware of anyone getting that carried away though. However, EVTV makes a 100Kw solar power system so it has to be using more than 14 modules. My first guess, knowing nothing of your set up, is that you might perhaps need more power. In the car the modules were powered by 5V. You can technically power them at 3.3V too but either way you need sufficient power. If you setup does not include a fairly powerful DC power source then you could run into issues. I doubt that an Arduino Due or other development board has enough power output to run a ton of modules. I think we tested up to 8 or so with just the development board but much over that you need a dedicated DC/DC power supply. So, let me know if you're using a sufficient DC/DC converter and if so we'd have to look elsewhere for the problem.

BobbyBleacher, 2019-12-03:
I'll go try the 5V side. It's true I've been using only the 3.3v output from the Due.

BobbyBleacher, 2019-12-03:
Ehh, unfortunately, no. Anything more than 14 modules and I get zero to show up.

collin80, 2019-12-03:
The Due is not a world renowned source of DC voltage. ;) Can you try with a benchtop supply or some sort of voltage supply that can provide 1A at 5V? Maybe a wall adapter? At some point you will overcome the current sourcing capacity of the Due and then you'll get weird problems. I can't say for 100% sure that's your problem but it could be ruled out with a separate supply.

BobbyBleacher, 2019-12-03:
I got it! It was an misunderstanding my part, actually. Your diagram is a big help, but the Tx/Rx lines from the left half of the connector go into the right half of the next connector. I had wired them straight over (I was adding these two modules to a factory wiring harness). It all works now. I'll close it. 

Thanks again for the software, Collin. I've edited it quite a bit now to mimic the EVTV setup and it seems to be working fairly well. Check out my fork some time!

closed 2019-12-03 by BobbyBleacher