collin80/TeslaBMS #12: Can't compile

taraskarpuk, 2020-01-16:
`Documents/Arduino/libraries/due_can-master/src/due_wire.h:25:25: fatal error: include/twi.h: No such file or directory`

Anybody know how to solve issue?

collin80, 2020-01-16:
Are you sure you're compiling for the Arduino Due? It's super easy to accidentally select the wrong board. The Arduino IDE is not super great at letting you know that you've got it set wrong.

taraskarpuk, 2020-01-17:
Yes, builded for Due it's fine. But i have fail at startup 
`Started serial interface to BMS.
Resting to factory defaults`

collin80, 2020-01-17:
The Arduino Due does not have EEPROM. That has to be added by whatever shield you're using (if you had one). Otherwise it will try to access EEPROM and not find one and freeze for a while as it tries to access the nonexistent chip. That's why it says it is resetting to factory defaults. It can't find valid EEPROM settings so it tries to write them to EEPROM to initialize. It won't be able to do so and so it freezes for a long time. The solution is to either add an EEPROM chip on the I2C pins of the Due or comment out all the places where it tries to use it and just leave things hard coded. That's what most people do.

taraskarpuk, 2020-01-17:
what baud rate is required for 1028578-01-C REV03 and shoud be yellow and blue shorted?

taraskarpuk, 2020-01-17:
`19:25:22.079 -> Sending: 7F 20 FF 7D
19:25:22.079 -> Reply: 7F 20 FF 7D 
19:25:22.079 -> Sending: 7F 20 0 8E
19:25:22.079 -> Reply: 7F 20 0 8E 
19:25:22.079 -> Sending: 7F 21 FF 68
19:25:22.079 -> Reply: 7F 21 FF 68 
19:25:22.079 -> Sending: 7F 21 0 9B
19:25:22.079 -> Reply: 7F 21 0 9B 
`

taraskarpuk, 2020-01-18:
Successfully connected, problem was in baud rate

closed 2020-01-18 by taraskarpuk