collin80/TeslaBMS #14: The problem with the code (arduino)

Bezzeb, 2020-05-31:
Good afternoon.
Thanks for the project and the work done.
I have a problem with the code for arduino.
I completely copied the code for arduino due,uploaded to the board
Connected arduino due to the tesla bms
Arduino due - level converter- tesla bms
+5 v                 HV                       Pin1        
                          +4.7KOhm         Pin6
gnd                  GND                    Pin3
3,3v                  LV  
TX                    Lv1                       Pin2
RX                    Lv2                       Pin9
The bms LED lights up when turned on, and then flashes briefly
on TX and RX 3.3 volts

And there are a few problems
1)In the serial monitor receives information only after the upload code to the board
when reconnecting, nothing is displayed
2)after upload code to the board all information stops at "Resetting to factory defaults"
14:54:00.247 -> Starting up!
14:54:00.247 -> Started serial interface to BMS.
14:54:01.516 -> Resetting to factory defaults

I understand this is the problem of arduino due ?
not tesla bms ?

collin80, 2020-06-02:
Did you try sending a ? with a line ending (that is, push ENTER after ? or have the Arduino Serial terminal send ? and tell it to use some sort of line ending). That will give you a menu where you can select what you want to do. If that's not working then something has locked up and we'd need to figure out where and why..

Bezzeb, 2020-06-16:
> Did you try sending a ? with a line ending (that is, push ENTER after ? or have the Arduino Serial terminal send ? and tell it to use some sort of line ending). That will give you a menu where you [...]

Thank you so much
Everything works fine
Is it possible to get data on all battery cells? (voltage)

closed 2020-06-28 by Bezzeb