collin80/TeslaBMS #15: Problem getting code to compile

ahrenswett, 2020-06-23:
@collin80 HI! First I want to say thank a ton for taking this on! Its invaluable and so glad I can use the existing board.

I am new to Arduino and C++ usually code in java so hopping you can help me get this going!
here is a link to the commit on my fork with the changes I made for my Due. [My commit](https://github.com/ahrenswett/TeslaBMS/commit/bddc6d23be85af160ee13dc2fdeb8958de1c6492)

I was able to get all the libraries you wrote included as that was first issue. Then I ran into this and changed from serial to USART that seemed to work or at least not error out
```
//Set to the proper port for your USB connection - SerialUSB on Due (Native) or Serial for Due (Programming) or Teensy
#define SERIALCONSOLE   Serial

//Define this to be the serial port the Tesla BMS modules are connected to.
//On the Due you need to use a USART port (Serial1, Serial2, Serial3) and update the call to serialSpecialInit if not Serial1
#define USART  Serial2
```
But now it is throwing an `error
```
In file included from sketch\BMSModule.cpp:3:0:
sketch\BMSUtil.h: In static member function 'static void BMSUtil::sendData(uint8_t*, uint8_t, bool)':
BMSUtil.h:37:16: error: request for member 'write' in '0', which is of non-class type 'int'
         SERIAL.write(addrByte);
                ^
```
Again thanks a ton for all your work!

ahrenswett, 2020-06-24:
Realized my error in changing to:
`#define USART  Serial2`
changed to
`#define SERIAL  Serial1`
then changed the call to serialSpecialInit in TeslaBMS to:
`serialSpecialInit(USART1, BMS_BAUD)`

Now top error has to do with `externa EEPROMSettings settings;`

ahrenswett, 2020-06-24:
now im having issues with due_wire. where do I find that?
```
// Include Atmel CMSIS driver
#include <include/twi.h>
```

collin80, 2020-06-24:
https://github.com/collin80/due_wire

ahrenswett, 2020-06-24:
> https://github.com/collin80/due_wire
got that :) but had some other issues after ide reverted to uno board
here are my errors now. really struggling with this :( I pushed my changes just in case you can take a look.
feels like i'm going in circles. 

In file included from sketch\Logger.h:31:0,
                 from sketch\Logger.cpp:27:
sketch\config.h:10:0: warning: "SERIAL" redefined [enabled by default]
```
 #define SERIAL  Serial1
 ^
In file included from C:\Users\ahren\Documents\ArduinoData\packages\arduino\hardware\sam\1.6.12\cores\arduino/Arduino.h:44:0,
                 from sketch\Logger.h:30,
                 from sketch\Logger.cpp:27:
C:\Users\ahren\Documents\ArduinoData\packages\arduino\hardware\sam\1.6.12\cores\arduino/wiring_constants.h:43:0: note: this is the location of the previous definition
 #define SERIAL  0x0
 ^
```

ahrenswett, 2020-06-25:
> https://github.com/collin80/due_wire
Got it. is it normal to get red messages about redefining SERIAL and than have it compile?

collin80, 2020-06-25:
I don't seem to get that warning when compiling the code. I don't know where that's coming from. But, hopefully it works anyway.

ahrenswett, 2020-06-25:
> I don't seem to get that warning when compiling the code. I don't know where that's coming from. But, hopefully it works anyway.
Got It! notice the light flashing as i was checking ports looks like the pins on the end of my wires for my breadboard were to thin to give a good connection on the header to the arduino. wiggled a bit and now its golden! got the menu after typing ?

closed 2020-06-25 by ahrenswett