collin80/TeslaBMS #16: Raspberry Pi instead of Arduino?

thezeroth, 2020-08-13:
What does it take to run this code on a Raspberry Pi instead of Arduino? Raspberry Pi runs off 5v DC and also has IO pins with 5v DC which seems to be the same voltage that the Tesla BMS is using.

collin80, 2020-08-13:
Really comm to the Tesla BMB boards is TTL serial at 5V. So, if you can connect the RPi over serial you should be set. The baud rate is quite odd but one has to presume that the RPi can generate it. There are other pins that have fault signals and such but what the BMB needs to work is power and serial. The other pins you can ignore if you want. Or, you can connect them, they're on/off digital signals.

pilotniq, 2024-03-10:
FWIW, I'm running a Raspberry Pi talking to the battery modules BMS boards over a serial port (on my electrified boat). My Javascript (node.js) code is here: https://github.com/pilotniq/MJoulnir/tree/main/RaspberryPi/software/bms

thezeroth, 2024-03-10:
@pilotniq thanks for sharing, I will look into your code,  I was thinking about making "Node Red" on a raspberry pi to talk to Tesla BMS, That can open many possibilities  like notification via MQTT,  integration with node red web user interface to show battery state on web widgets, logging battery data on a database like InfluxDB, then visualizing that data via Grafana, etc. Last time I checked there is a CAN addon for Node Red.
Can you please also share what type of hardware you are using to interface pi to CAN and what model of raspberry pi? How did you wire all battery modules to the pi? Would be great if you could provide pictures of your setup.