collin80/TeslaBMS #17: Couldn't find a Board on the selected port.

49thdiver, 2020-11-22:
Greetings, I have tried to get this to go 3 or 4 times now with no success. I am not new to arduino programing but not an expert either. I have successfully built projects including can bus on uno, due & 2560 boards. I have done some can bus bridge work and am using ide 1.8.8
Your code complies and uploads via the native port, when the board is reset it can not find the serial port. The console on the programming port shows nothing. I do not have the Tesla BMB even connected. If I could get the "Starting up!" message I am sure I could debug the rest. 
From what I read elsewhere this error message is as a result of something wrong with the sketch corrupting the serial I/O or possibly the libraries are not compatible.
If I reset communications and reset the board I can get the serial ports back and compile and load known working sketches.
I dumped the clone boards and have the original Arduino product still no joy.
Collin I know this project is kinda dated and you may not have time for it but I need help. I have two Tesla modules I would love to be able to use with this code for a solar project. I would not bother anyone unless I had run out of ideas, but now I am now desperate. Error file attached, perhaps there is an indication there of an incompatible library. Thank you in advance !!
[teslabms error.txt](https://github.com/collin80/TeslaBMS/files/5579051/teslabms.error.txt)

collin80, 2020-11-22:
Looking at the error file you sent, it appears you have your libraries all messed up. TeslaBMS should not be in the libraries folder. That's the main program and it should be found elsewhere, basically anywhere but the libraries folder. It looks like maybe you have due_wire and wire_eeprom in a folder called kidder which I suppose isn't a problem so much about the folder name. But, those are two separate libraries and you seem to have them in the same folder. That probably isn't really the problem either but it isn't really "correct." I think you might want to delete the libraries and start over. Download the actual libraries one at at time and put them in your libraries folder. Then put TeslaBMS somewhere else, say in your Documents folder, open it, and try compiling. I don't know why you are able to compile but it doesn't work when it has uploaded. But, I do hope that cleaning up your libraries might fix it.

49thdiver, 2020-11-22:
Thank you for your quick response, all I can say is "Remarkable". Last night  after posting my call for help I decided to get dramatic and set to clean up my libraries by removing all of the unnecessary folders. I then started to remark out major sections of the code and the includes to cut it down to something that at least created some output on the programming port console using Serial.print. At around midnight I was able to get output on the programming port but still noting on the "SERIALCONSOLE". 90% of the code was eliminated. Introducing line by line the #includes caused it to fail again.
Went to bed.
You email this morning was a welcome conformation I was on the right track, not sure how TeslaBMS got stuck in the library, that was a result of rebuilding 1/2 a dozen times and getting sloppy. So I set to implement your suggestions, significantly, removing duplicate can_due files, placing wire and eeprom in the own directories with appropriate names and moving TeslaBMS to the appropriate place in a sub-directory of my Arduino folder with all my other projects directories.
This has not completely resolved the situation but is significantly better. It Compiles and loads successfully, the only out put I can get however is via Serial.print.
Things to note are that the upload process now ends with the ide printing "Done uploading" on the status bar all the time and more significantly the Port on the Tools menus is correctly identified by name not just a port number, which never appeared to be correct, i.e. com15 not com16 where the correct named port is.
I am now working thru the code to determine why I am not getting any output on the "SERIALCONSOLE", I have been inserting "Serial.print" commands to step thru the code. The SerialConsole.h is included at the start.
I am not sure how the logger is supposed to work in the arduino world I have used a similar tool in the android environment, but I am only using the Arduino ide for this work so far.
I do have Atmel Studio 7 and an AVR that I am using for another ATxmega32C4 project but have not used it for any of my other arduino projects yet. Should I be using that and will it help to debug. I anticipate I will find my way thru this now thanks to your wisdom. If I am missing something else obvious please let me know. I have not looked at the "SerialConsole.h" or "Logger.h" files yet perhaps all will be revealed.

Thanks for taking the time to read thru my error log and point me in the right direction.
All the best to you, Peter

[Nov22_TeslaBMS_Log.txt](https://github.com/collin80/TeslaBMS/files/5580173/Nov22_TeslaBMS_Log.txt)

49thdiver, 2020-11-22:
AH ha, clearly should have looked at the config.h file associated with SerialConsole.h and Logger.h
I needed to change from the default "#define SERIALCONSOLE   SerialUSB" to "#define SERIALCONSOLE   Serial"
It appears to now be ready to connect to the batteries, I have the level converters and cabling ready to go.
Thanks, Still learning ............ after all these years ;).
Will close file after testing with batteries.
Thanks Collin.

49thdiver, 2020-11-24:
I have had some success, with good data coming over from the battery, however I am mostly I am getting bad data, back from the batteries. I have used baud rates between 608695, 612500, 617647and 631578 it all get s confusing but I think my good results were obtained at 612500 or 617647. It may be I am having some hardware issues but your software is for sure working. The key to getting it workings was cleaning up the library and changing the config.h file for my configuration. Thanks for you work in developing this and sharing it. It's your generosity and others that help people like me develop new idea's and project and in turn share what we have learned. Thank to everyone. I will close this comment and drop a note here when I get the hardware working.

closed 2020-11-24 by 49thdiver

evcepi, 2025-03-22:
Hello 
Maybe help
I not 

[image]
[image]
[image]
[image]

see voltages and cells?