collin80/TeslaBMS #18: Tesla BMB heating

sagez400, 2021-05-17:
Hello Collin,
I am working on a college research project involving the use of Tesla Model S batteries. 
Following the previous issues discussed, I was able to get the code working. 

However, I ran into an issue of the boards heating up a lot (in the area between the cell connectors) when module information was being displayed to the serial monitor. Regardless, the data is still being read properly.

Setup:
Tesla Model S P100D battery modified for 48V
Two Tesla BMBs (one for cell group 0-5, and one for cell group 6-11) 
Boards run on 5V from the Arduino Due, powered by each proper group of cells as mentioned above, and use RX1/TX1
Only one temperature sensor is connected since I am only using one module
Baudrate set at 631578

I'm not sure if the code or the 5V from the Due is a cause here.
Any suggestions as to where I could start troubleshooting the heating problem?

collin80, 2021-05-17:
Are you sure something isn't commanding balancing? That will heat up the cells.

sagez400, 2021-05-18:
I did not notice at first that the void loop function had an interval balancing command. 
Heating issue is solved, thank you!

closed 2021-05-18 by sagez400