collin80/TeslaBMS #20: ERROR: Invalid module response

Bernhard1023, 2022-01-16:
I have 4 Tesla Battery Packs (Model S 2012), which I tried to read out the BMS with an Arduino Due and a level shifter.
(https://www.az-delivery.de/products/logiklevel-wandler-3-3v-5v?variant=18282761715808)
I always get this error, and I don't know how to proceed further...
The error:
5447 - ERROR: Invalid module response received for module 1  len: 3   crc: 8   calc: 45
6448 - ERROR: Invalid module response received for module 1  len: 3   crc: 8   calc: 45
7449 - ERROR: Invalid module response received for module 1  len: 2   crc: 8   calc: 14
8450 - ERROR: Invalid module response received for module 1  len: 3   crc: 8   calc: 45
9451 - ERROR: Invalid module response received for module 1  len: 3   crc: 8   calc: 45
...
This occurs at all 4 modules.
Is the communication broken, or does it even happen?
Where is the best point to search for the problem?

collin80, 2022-01-17:
Hmmm, well, the fact that the CRC calculation is showing a mismatch probably means that the traffic is being messed up on the wire. Sometimes level shifters can be problematic. On the "official" EVTV board we used AND gates as a buffer. Alsom afterward it became known that you do not need to level shift at all. The Tesla modules work fine at 3.3v so if you power them via 3.3v you can directly interface them to the Due without any additional hardware in between. This is probably the easiest path forward.

Bernhard1023, 2022-01-18:
Thanks for the fast response.
I removed the level shifter and tried it out with the 3.3v.
Right now my serial output looks like the following:

Starting up!
Started serial interface to BMS.
Resetting to factory defaults

If I unplug the Tx wire, I get similar errors as described before, just with a lot of zeros, which makes sense for me.
Why do I get no output? (I use the Programming Port on the due and changed the line in the config.h to "Serial" as  described.)

It seems like the software is entering the main loop because the LED on the Tesla Module is flashing every second, which should be the indication for a data transfer, or at least the try to transfer.