collin80/TeslaBMS #22: Tesla BMS test

KGog, 2022-05-31:
Hello 
After uplode I am getting such error mesage Arduino: 1.8.19 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\kgogoladze\Documents\Arduino\Tesla_BMS\BMSModuleManager.cpp:2:0:

BMSModuleManager.h:4:10: fatal error: due_can.h: No such file or directory

 #include <due_can.h>

          ^~~~~~~~~~~

compilation terminated.

exit status 1

due_can.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

collin80, 2022-05-31:
That's because you are compiling for an ATMEGA board while the code is meant to be compiled for the Arduino Due. It's a completely different processor. You will also need the due_can library so ensure you've installed that as well. I wrote the due_can library so you can find it in my repos here.

KGog, 2022-06-20:
Hello,
Now I tried with the DUE board but without any success, the same error is presented. Could you please indicate where I can find due_can.h file?
Thank you in advance
Looking forward to your reply

In file included from C:\Users\kgogoladze\Documents\Arduino\Tesla_BMS\BMSModuleManager.cpp:2:0:
C:\Users\kgogoladze\Documents\Arduino\Tesla_BMS\BMSModuleManager.h:4:21: fatal error: due_can.h: No such file or directory
 #include <due_can.h>
                     ^
compilation terminated.

Compilation error: exit status 1

KGog, 2022-07-30:
Hello,
I think I have success I have run your code
Could you please check atteched log file? 
[LOG.txt](https://github.com/collin80/TeslaBMS/files/9227142/LOG.txt)

collin80, 2022-07-31:
That looks like a successful build to me

ViktorGys, 2025-04-10:
Здраствуйте! Такая же проблема.
 due_can.h: Нет такого файла или каталога.
Что делать?