collin80/TeslaBMS #24: Native USB and reset button goes dead

daimlerman, 2022-10-03:
Hello,
I am working on a project were i want to talk to some older tesla battery modules for  a home battery. Battery module are out of  a MB B class and sketch is uploaded to a Arduino Due
My sketch is somehow disabling the reset button and the native USB port. the only way to get it back is to erase the Due. Once erased the USB port is recognized by the PC right away and the reset button functions again. 
I configured the config.h to #define SERIALCONSOLE   SerialUSB.
The sketch compiles with out error and a second Due acts the same way.
What other setting am I missing?

Many thanks

twanek, 2022-12-02:
Hey!

If it is still relevant, try to use the other micro usb port, not the native. To switch, in the code just edit `#define SERIALCONSOLE SerialUSB` to `#define SERIALCONSOLE Serial`, then in the IDE also select programming port.

daimlerman, 2022-12-04:
Thank you Twanek for responding! 
I had figured this out and fix by reorganizing the depository structure, much like a post from the closed issues section. 
Besides using the programming port, i have yet to get a reliable communications going. I reports the wrong numbers of modules and the voltages are all over the map. 
Has you or anyone else nailed the baud rate down for the B-class module?

twanek, 2022-12-05:
Are you sure your module is 6s and not 7s? For me, with the 7s module (Merc C class), it has issue correctly identifying the number of modules, and the voltages too. See my opened issue here: https://github.com/collin80/TeslaBMS/issues/26

As for the baud rate, the C class module seemed to work with most of the baud rates specified in the sketch. Did you tried the 631578 baud?

collin80, 2022-12-05:
I've got to be honest, I've never heard of a system using this chip that has 7S modules instead of 6S. So, this is new to me. However, this much I can tell you - the BMS chip used supports 3 to 6 cells. I'm sure you can see where this would be problematic since it cannot read 7 cells. The only viable solution is thus to have 3 cells on one chip and 4 on a second chip. This makes each module look like 2 modules to the TeslaBMS code because there literally are two chips it is talking to. You're going to have some trouble here. In theory, there's no reason that all the code could not be modified to expect 3 or 4 cells per BMS chip but this will require a lot of code changes. And, unfortunately, I'm not going to be able to do this myself because there is such a small market for these oddball modules you're using. So, you'd have to modify the code to expect 3 and 4 and combine them into 7S modules so you can keep track of them properly.

twanek, 2022-12-06:
Thanks for the explanation, this makes thing more clear to me. So, the fact that it detects 2 modules is not a problem at all, this is expected behaviour in this case? Regarding the code adaptation for 7s, I do not think this is big problem, I will use it with a known number of modules and will hardcode all the chip id pairs in the program, so it adds up all 7 cells from the module, etc.

I have few more questions :)

1) just for curiosity, the "standard" 6S modules do have only one chip?

2) what communication protocol is actually used between the Due and the BMS? Sorry if this is a noob question, but:
- if CAN protocol, then why use the strange baud rate and specific TX/RX pins?
- if Serial protocol, then why the CAN library is needed?
this is confusing for me...

3) what about the 2 "extra" temperature readings? The BMS has only 2 temp sensors, but both chip sends 2 different values, totally 4 temperatures? Which one is the real one? It could be that the communication protocol just reads some "extra" addresses from the second chip which is not meant to read? Please take a look at my log file here:
https://github.com/collin80/TeslaBMS/files/10148710/log.for.bms.v02.7s.617647baud.txt

Actually, the 7S modules are very useful for motorhome solar batteries, because with 7s lithium configuration you get a 24v alternator "compatible" pack... My truck alternator outputs 28.8v, so it charges to max 4.11v each cell, which is just perfect. I have (4 tesla modules in parallel configuration).

This way I could connect the starter battery and tesla battery pack together (via a huge current limiting resistor, to protect the alternator not to be shorted by tesla pack). This system works perfectly since 2 years, but I would like to be able to use the built in BMS on the modules.

Thanks!
best regards,

daimlerman, 2023-01-10:
Thank you for all the helpful information. I've got the setup mostly to connect and run reliable. 
Collin, you are a Champ!

closed 2023-01-10 by daimlerman