collin80/TeslaBMS #25: Random Invalid module response received

Jcr787, 2022-10-16:
Im having these random errors. Errors are not module constant. Im trying to isolate the problem but Im out of Ideas at this point. 
I tried changing comms from serial 1 to serial 2 but problem persists. Do you have any idea what could cause this? Here are some examples 
2195768 - ERROR: Invalid module response received for module 9  len: 22   crc: 130   calc: 138
2198530 - ERROR: Invalid module response received for module 10  len: 22   crc: 166   calc: 205
2198551 - ERROR: Invalid module response received for module 1  len: 22   crc: 67   calc: 5

It happens more frequent when Im running my LF inverter, not sure if EMI might be the culprit. 
Im using a arduino due with direct 3.3v connection, also tried 5v with an bi-directional logic level converter but still get random errors.
I added contactor for precharge and main. Contactors and arduino are powered with a 48v to 12v stepdown converter. 

Thanks in advance for your asssitance.

collin80, 2022-10-18:
Yes, it's probably EMI. You're getting the proper number of bytes returned but not the proper CRC so chances are the data has been corrupted. If this only happens every so often you can essentially ignore it. The code knows to discard any traffic that was corrupted so it shouldn't mess anything up. It just means you aren't always getting the latest data.

About all I can suggest is to keep your noisy wires away from the signal wires, if they have to cross it is better to do so perpendicularly, and perhaps shield and/or twist your noisy power wires.

daimlerman, 2022-12-06:
Thanks for responding,  I am having the exact same issue where my 10 boards show up as 20 with different voltage levels.
When combining the displayed voltages for two boards i.e. module 1&2 that would reflect the correct voltage for actual board one. 
My setup has 2 groups of  5 parallel modules in series for a 60V system. The problem arising is that the two groups of modules drift apart in voltages. There seems to be a .3 .to .5 of volt difference. How do I balance that? I attached a snippet of a log. 
[Info level log.txt](https://github.com/collin80/TeslaBMS/files/10162715/Info.level.log.txt)

Jcr787, 2023-10-11:
Im still dealing with this issue. The lf inverter mosfet switching is causing a ringing noise that get sent thru the battery cables to the modules slave bms. This gets to the UART wires. @collin80 how have you dealt with this before?

collin80, 2023-10-12:
Ah yes, a problem near and dear to all of our hearts. Inverter switching noise is a nasty thing to deal with. There are multiple layers at play here. The Model S modules regenerate traffic at every hop. So, comm becomes clean again at every module and is re-sent fresh to the next in line. This limits the amount of noise that can be coupled. So, the best advice for the wiring near the batteries is to keep it away from your power wires as much as possible. When they come near each other they should do so at right angles if possible. This should limit the amount of noise and every module refreshes the signal so noise is not cumulative.

Long runs of wire would be best if you could shield them or shield the power wires. If at all possible they should not run beside each other parallel. If you have to do that then one or the other or both should be shielded. Ground the shield at one end only to make it into a big antenna to ground. Lastly, the modules run fine at 3.3v but if you can utilize 5v instead you will be better off. The higher the voltage differential between states, the harder it is to corrupt. The official EVTV boards all use 5V signaling down the lines. You can use 3.3v and people have. But, that will be a bit more noise prone. I believe the lowest voltage that works is around 2.7v.