collin80/TeslaBMS #26: How to change number of cells?

twanek, 2022-12-02:
Hi!

I managed to get some output with a Due, but my modules are S7, not S6  (it is from Mercedes C class).
The problem is, that in your version the number of cells are only up to 6, so I get wrong values for AVG voltages and for one cell voltage it always  reads 0.0V.

Can you please advise, what is needed to change in the code to read all 7 cells? It is enough to change all the "for cycles" to 7, or it needs some additional hex address, etc in the protocol?

Thank you,
best regards!

twanek, 2022-12-04:
Hi again!

I made some more tests, printing detailed info about the module, and there is some strange thing going on:

I have **only one module connected**, but the Due **always finds 2 modules**. Regarding the cell voltages, on the "Pack status" it prints for the "first" module 4 correct values for the cells, then for the "second" module it prints 3 correct values for the rest of the 7 cells of the first module...

Also, for the temperature, AFAIK, the module has only 2 temp sensors, but it prints 4 different temp value, 2 for every "module"...
Attached there is a log file and few photos about the BMS, the version number is readable, so you can understand better what I'm trying to explain. There is only 1 module connected!!!

[log for bms v02, 7s, 617647baud.txt](https://github.com/collin80/TeslaBMS/files/10148710/log.for.bms.v02.7s.617647baud.txt)
[image]
[image]

Can you please look into the log file, what can be the issue, why it finds 2 modules instead of 1?

Thank you!
best regards,

daimlerman, 2022-12-10:
The module in the pictures are definitely from a MB B- class. 
Manufacturer module specs are; 2.3KWh of usable energy content; low voltage limit 17.5V; high voltage limit 29V; 308 cells
I am not sure what chemical composition for the cell is, but I would not discharge them below 2.8V (19.6V for the module).

twanek, 2022-12-11:
Thanks for the info!

Indeed has 308 cells, it is a 7s module. I never discharge it under 3.0v, usually 3.1v. Anyway, under 3.1v there is nearly no energy at all, the voltage drops very quickly. We use 4 modules like this in parallel, to power our motorhome.

mart1npetroff, 2025-03-17:
Hello @twanek @collin80 I'm plannig to use this for testing 7s B-class battery as well.
Could you provice me more information if  I need to do any code changes and how to deal with the four values for the two temp sensors?
If you just give me any guidelines I could do the changes and provice PR with the changes.
Thank you.

twanek, 2025-03-17:
Hi!

Its been a while since I worked on this project. Unfortunately, I had to pause the progress, because with 2 kids it is hardly any free-time :) However, I still would like to finish it. Give me few days, will try to review my code, then will share my ideas with you.

twanek, 2025-03-23:
Hi!

My use scenario is 4 modules built in my motorhome, so it is a static environment. For this, my approach would be to hard code the batteryID of each module (actually 2 ID for one module, because there are 2 BMS chips on each module: one reads 4 cells, and one reads 3 cells. Also, the same is for the temp sensors, there are dummy values, you must keep only the correct ones.)

So, for example the first module would consist of ID1 + ID2, second module ID3 + ID4, etc. (hardcoded, thus the program knows which two ID's make one physical module. You read out all the info from the can frame, but then ditching the values from unused cell numbers and temp sensors).

If you work with the same modules all the time like me (and probably most people), this approach should work for you too.

Hope this makes sense.